FleetProbe/MicroBridge-Arduino-ATMega2560 #1: Question: What happends when eeprom byte is set but file not exist?

Apollon77, 2015-07-21:
Hi,

can you tell me what happends when the bit it set but the file does not exist on sd card?

gerdmestdagh, 2015-07-21:
Hello,

It seems if there's no file the bootloader will check each time a reset is
triggered to see if the file is there, only when the file is found and
after the file has been put in program area then will the eeprom flag be
set to default and thus not check for the file again.

We are using a flash chip now to update remotely, there' s other branches
who support this.

Kind regards,

Gerd

Apollon77, 2015-07-21:
Thanks. I have a standard Arduino ... can I use the version with flash chip too? Which branch is it?

gerdmestdagh, 2015-07-21:
You need an external flash chip for this. This is a seperate chip which
will require some circuitry made by yourself. The only thing i can
recommend is look up the datasheet of such a chip and learn from it.

The bootloader only works for the mega2560.

Apollon77, 2015-07-21:
Ok, then i would go with sd card because i already have it on my ethernet shield :-) so version from master is correct one i assume ... or ?!
BTW: I found he branch with from-flash ... but there is no real example and also readyme is identically to sdcard one ... would help I think ...

gerdmestdagh, 2015-07-21:
Use master then, you might have to adjust the cs pin when you use the
ethernet shield

Apollon77, 2015-07-21:
CS for Sd card is pin 4 as I know ...

gerdmestdagh, 2015-07-21:
You should set the cs pin correctly in asmfunc.s. It is defined there under
the ethernet comment but i had changed it to work with our own shields.

Apollon77, 2015-07-21:
Can you tell me what would be correct for Mega2560 pin 4 ? In Pinout I found "PG5" for pin 4 ... I would assume the following then:
#define	DDR_CS	_SFR_IO_ADDR(DDRG), 5	// MMC CS pin (DDR, PORT)
#define	PORT_CS	_SFR_IO_ADDR(PORTG), 5

correct?!

gerdmestdagh, 2016-12-20:
PORTG 5 is digital pin 4 yes