FleetProbe/MicroBridge-Arduino-ATMega2560 #2: Error Loading Board in new Arduino IDE v1.6.5

Apollon77, 2015-07-21:
I have copied the "hardware"-Filder to my Arduino dir and restarted IDE (v1.6.5). Now I got the following error in IDE as soon as I open the "Bard manager":

Could not find boards.txt in /Users/.../Arduino/hardware/microbridge/bootloaders. Is it pre-1.5?
WARNING: Error loading hardware folder /Users/...Arduino/hardware/microbridge
  No valid hardware definitions found in folder microbridge.
Could not find boards.txt in /Users/.../Arduino/hardware/microbridge/bootloaders. Is it pre-1.5?
WARNING: Error loading hardware folder /Users/.../Arduino/hardware/microbridge
  No valid hardware definitions found in folder microbridge.

Apollon77, 2015-07-21:
When I move the programmertxt and board.txt to the awaited dir according to error message the next warning is:

Board microbridge:bootloaders:microbridge doesn't define a 'build.board' preference. Auto-set to: BOOTLOADERS_MICROBRIDGE

gerdmestdagh, 2015-07-21:
It could be some things have changed in the newer version of arduino ide. I
was using 1.0.6

Apollon77, 2015-07-22:
I found that:http://www.lydiard.plus.com/chapter14.htm

gerdmestdagh, 2016-12-20:
new arduino not supported

closed 2016-12-20 by gerdmestdagh

j54n1n, 2018-12-08:
I have created a PR with the updated board definition in case someone needs this