FleetProbe/MicroBridge-Arduino-ATMega2560 #3: Compiler warnings when recompiling bootloader

Apollon77, 2015-07-22:
Hey, I needed to change the CS pin value and so I recompiled after installing CrossPack on my Mac.

I got the following output with warnings:

Compiling: stk500boot.c
avr-gcc -c -mmcu=atmega2560 -I. -gdwarf-2 -DF_CPU=16000000UL  -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -fno-jump-tables -mrelax -Wall -Wstrict-prototypes -Wa,-adhlns=stk500boot.lst  -std=gnu99 -MD -MP -MF .dep/stk500boot.o.d stk500boot.c -o stk500boot.o 
stk500boot.c: In function 'main':
stk500boot.c:1148:9: warning: passing argument 1 of '__eerd_byte_m2560' makes pointer from integer without a cast [enabled by default]
         if(eeprom_read_byte(0x1FF) == 0xF0){
         ^
In file included from /usr/local/CrossPack-AVR-20131216/avr/include/avr/boot.h:104:0,
                 from stk500boot.c:107:
/usr/local/CrossPack-AVR-20131216/avr/include/avr/eeprom.h:647:9: note: expected 'const uint8_t *' but argument is of type 'int'
 uint8_t eeprom_read_byte (const uint8_t *__p) __ATTR_PURE__;
         ^
stk500boot.c:1167:16: warning: passing argument 1 of '__eewr_byte_m2560' makes pointer from integer without a cast [enabled by default]
                eeprom_write_byte (0x1FF, 0xFF);
                ^
In file included from /usr/local/CrossPack-AVR-20131216/avr/include/avr/boot.h:104:0,
                 from stk500boot.c:107:
/usr/local/CrossPack-AVR-20131216/avr/include/avr/eeprom.h:674:6: note: expected 'uint8_t *' but argument is of type 'int'
 void eeprom_write_byte (uint8_t *__p, uint8_t __value);
      ^
stk500boot.c:1153:24: warning: unused variable 'i' [-Wunused-variable]
                uint8_t i = 0;
                        ^

Compiling: pff.c
avr-gcc -c -mmcu=atmega2560 -I. -gdwarf-2 -DF_CPU=16000000UL  -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -fno-jump-tables -mrelax -Wall -Wstrict-prototypes -Wa,-adhlns=pff.lst  -std=gnu99 -MD -MP -MF .dep/pff.o.d pff.c -o pff.o 
pff.c: In function 'get_fat':
pff.c:382:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   return LD_WORD(buf);
   ^
pff.c:386:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   return LD_DWORD(buf) & 0x0FFFFFFF;
   ^
pff.c:358:15: warning: unused variable 'ofs' [-Wunused-variable]
  WORD wc, bc, ofs;
               ^
pff.c:358:11: warning: unused variable 'bc' [-Wunused-variable]
  WORD wc, bc, ofs;
           ^
pff.c:358:7: warning: unused variable 'wc' [-Wunused-variable]
  WORD wc, bc, ofs;
       ^
pff.c: In function 'pf_mount':
pff.c:760:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     bsect = LD_DWORD(&buf[8]); /* Partition offset in LBA */
     ^

Apollon77, 2015-07-22:
PS: And bootloader "Program" is then a bit smaller then in your example: 
```
Size after:
AVR Memory Usage
----------------
Device: atmega2560

Program:    5864 bytes (2.2% Full)
(.text + .data + .bootloader)

Data:        333 bytes (4.1% Full)
(.data + .bss + .noinit)
```

Is this ok ?!

gerdmestdagh, 2015-07-22:
The warnings are normal, and 5800bytes is still ok you can not exceed 8kb
of bootloader area. If the pin is correctly mapped it should work.

Kind regards,

Gerd

closed 2018-12-11 by gerdmestdagh