FleetProbe/MicroBridge-Arduino-ATMega2560 #5: Not able to update from firmware.bin

brunoaduarte, 2016-06-28:
Hi,

I've formatted a SD Card with a 2 GB partition.
Put a "firmware.bin" file in root of SD.
Set EEPROM 0x1FF to 0xF0. (verified)
Activate watchdog and reset device (verified).
But device is never updated with the firmware.bin file.

I burned the MicroBridge bootloader using a Arduino UNO as a "Arduino ISP".
Pinout is ok (pins 50, 51, 52 and 53) (PORTB,3 -> PORTB,0).

Anything else i should do ?
No errors occurs and i can program my board like the regular mega2560 bootloader.

gerdmestdagh, 2016-06-29:
you should look to set the correct cs pins in the bootloader code. It
varies between different shields. which shield are you using ?

brunoaduarte, 2016-06-29:
Sorry @gerdmestdagh , i was using the wrong CS pin in code (schematic was a little confusing). I'm using a custom made board. Now it worked !! Thank you !

brunoaduarte, 2016-06-29:
@gerdmestdagh , now i have a problem with the "flashstartexample.ino". The `wdt_reset()` is not triggering the bootloader code (it jumps directly to the app again and never gets updated). It only works if i do a physical reset (firmware.bin is successfully loaded). Any ideas ?

closed 2016-06-29 by brunoaduarte

gerdmestdagh, 2016-07-01:
glad to hear you figured it out :)