FleetProbe/MicroBridge-Arduino-ATMega2560 #8: Question: Recompile the bootloader to use MegaCore instead of the default 2560 pinout?

theshadow124, 2018-06-23:
Hello,

I am not entirely sure where to begin doing this or if this is the right place to ask, I really want to try out your bootloader on a little project I am working on. But I needed to use a few extra pins that are not normally available on the Atmega2560 via the standard Arduino pinout so I am using the MegaCore pinout, located here: https://github.com/MCUdude/MegaCore

Is there a way to change the target board for the bootloader while I compile it so I can use the other pinouts/interrupts/ A-Ds that are not normally available?

Thanks,
Kyle

gerdmestdagh, 2018-06-23:
Hi,

It's been a while since I made this project. Mechanics are :

   1. Sketch is detected on SD-Card
   2. My modified bootloader picks this up and writes it to flash memory

From what I see Megacore has its own bootloader. I don't know if you need
to use this bootloader or if you can just use a arduino bootloader ( in
which case you could write the Mircobridge bootloader )

Kind regards,

Gerd

theshadow124, 2018-06-23:
Hello Gerd,

Thank you for your response. My understanding of the Megacore is that it is essentially the stock Arduino bootloader giving access to more of the pins of the ATMega2560 as well as a few other small configurations.

I want to use your bootloader, but I currently use quite a few pins that are not mapped out by the default 2560 pinout.

Kyle

gerdmestdagh, 2018-06-25:
Then you are out of luck :) You can look what I adjusted in the bootloader
and try to implement this in the Megacore bootloader that's one option.
That or access your pins with bit manipulation in your sketch (the
registers). You should be able to use these pins without the megacore
bootloader

Kind regards,

Gerd

theshadow124, 2018-06-25:
Hello Gred,

The MegaCore makes a lot of the extra pins significantly easier to use, and all of my code already relies on it. I'll see if I can modify the MegaCore bootloader to include the SD self-flashing functionality.

Do you have any insight to share to make my life a little easier? If not just close this and I will try it.

Thanks,
Kyle

gerdmestdagh, 2018-06-26:
Hello Kyle,

Not really, get an avrtool for quick bootloader flashing. It will help a
lot as opposed to using another arduino as flasher.

Kind regards,

Gerd