FleetProbe/MicroBridge-Arduino-ATMega2560 #10: Suck on boot, LED flashes on for 6 second soff for half a second

kawasakizx10rr, 2018-08-24:
Hello I have flashed the bootloader to my mega2560 and when i set the eeprom to 0xF0 and reset the mega the LED blinks on for 6 secs and off for a half a second and it never boots to the code, unless I unplug the SD card OR set the eeprom back to 0xFF ?? Any Idea what's going on please.

I made a new board and flash the bootloader as i describe below.
 
C:\Users\Myname\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.17 

And a new bootloader folder named microbridge in the bootloader folder of the same directory copying the entire folder contents 
bootloader files
###############################################################

microbridge.name=Arduino Mega2560 Dual SD Boot

microbridge.vid.0=0x2341
microbridge.pid.0=0x0010
microbridge.vid.1=0x2341
microbridge.pid.1=0x0042
microbridge.vid.2=0x2A03
microbridge.pid.2=0x0010
microbridge.vid.3=0x2A03
microbridge.pid.3=0x0042
microbridge.vid.4=0x2341
microbridge.pid.4=0x0210
microbridge.vid.5=0x2341
microbridge.pid.5=0x0242

microbridge.upload.tool=avrdude
microbridge.upload.protocol=stk500v2
microbridge.upload.maximum_size=258048
microbridge.upload.maximum_data_size=8192
microbridge.upload.speed=115200

microbridge.bootloader.tool=avrdude
microbridge.bootloader.low_fuses=0xFF
microbridge.bootloader.high_fuses=0xD8
microbridge.bootloader.extended_fuses=0xFD
microbridge.bootloader.file=microbridge/stk500boot.hex
microbridge.bootloader.unlock_bits=0x3F
microbridge.bootloader.lock_bits=0x0F

microbridge.build.mcu=atmega2560
microbridge.build.f_cpu=16000000L
microbridge.build.board=AVR_MICROBRIDGE
microbridge.build.core=arduino
microbridge.build.variant=mega

and then made the binary file as you describe

avr-objcopy.exe -I elf32-avr -O binary Blink.ino.elf firmware.bin

then placed that firmware.bin in my SD card.

I have tried both FAT and FAT32 formats and both formats work fine with the standard SD library and i am able to see the firmware.bin file

I then use you firstRun code to set the EEPROM block EEPROM.write(0x1FF,0xF0); and it resets and then just hangs on a blinking LED on for about 6 seconds and off for about half a second, I assume it it constantly trying to load the code from the SD card ??

I have not touched the stk500boot.hex file as some are rebuilding it, do i need to modify it ??
I am using the standard pins 50,51,52,53 with a 2Gb SD card which again works fine with the default SD card library.

Please help this is for a company and i can pay you for a working example.

gerdmestdagh, 2018-08-24:
you can easily build the bootloader yourself. You just need to set the
correct SPI Pins for your device that is serving the bin file.

Kind regards,

Gerd

kawasakizx10rr, 2018-08-24:
I used the pins 50,51,52,53 as referred to here https://www.arduino.cc/en/Reference/SPI

Please please for the love of god help, i have no idea how to build the bootloader, can i not just pay you to make it quickly, via paypal or whatever. please

kawasakizx10rr, 2018-08-24:
Can I pay you for a working version if this bootloader, whatever price you think is reasonable. Please.

closed 2018-12-11 by gerdmestdagh