FleetProbe/MicroBridge-Arduino-ATMega2560 #12: Can't change pins

kernel64bits, 2019-01-24:
Hello guys, 
Due to technical reasons I cannot use the original pins, instead I'm using pins 46, 47, 48 (PORTL) and 7 (PORTH). Unfortunately the addresses of the registers linked to these ports are way beyond 32 so I cannot use the functions sbi, sbic, and chi (which are usable only for addresses between 0 and 31). Instead I was thinking of using sbr, sbrc, and cbr.

For example I replaced
`sbi	DDR_CS		; CS: output`  
by
 `lds r16,myCS_DDR` 
`sbr r16,myCS_BIT`
`sts myCS_DDR,r16`

Since sbr behaves in a slightly different way than sbi I had to define:
`#define myCS_DDR	DDRB`
`#define myCS_PORT	PORTB`
`#define myCS_BIT	1`
So far I'm just trying to get rid of the problematic instructions before trying to change pins.

This trick is working well for the first function (init_spi), unfortunately when I try to apply it in other functions, the bootloader is not working properly (the firmware update is not made and the board often even crash).

I joined my modified version of the code (I put the non-working modifications as comment).

[asmfunc.S.txt](https://github.com/FleetProbe/MicroBridge-Arduino-ATMega2560/files/2792217/asmfunc.S.txt)

Thanks in advance

kernel64bits, 2019-01-24:
I also tried replacing r16 but it doesn't work better

gerdmestdagh, 2019-01-24:
It's been a while since I have used or even look at this code. Have other
work now ...
But can you tell me which board you are using ?

kernel64bits, 2019-01-24:
I'm using Arduino Mega 2560

kernel64bits, 2019-01-25:
Yup, that's what I did but my modifications of the code to replace sbi, sbic and cbi are not working. So far I'm just trying to use the default pins without these instructions (so I can replace only 1 instruction at a time and see if it's working, that's how I realised that modifications on init_spi are working but not the others). Then I assume I'll be able to change the pins as you showed.

> But you need the spi interface on the chip and those are on pins d50-d53

What do you mean exactly ? My aim is to replace these pins

gerdmestdagh, 2019-01-25:
the atmega 2560 has miso mosi sck pins defined for the spi interface. This
is an interface for talking to spi devices such as memory. The only thing
you can change in the spi lines is SS( slave select) which selects which
spi device you are talking to. Look at the pinout of the atmega you will
see what I mean. I am not 100% sure but I think you will have to use the
50-53 for this because those are the spi lines.

kernel64bits, 2019-01-27:
> The only thing you can change in the spi lines is SS( slave select) 
Ok, I'll try to change to change another pin to an address between 0 and 31

gerdmestdagh, 2019-01-28:
> I managed to change SCK to port 28 (A6) by just changing
> #define DDR_CK _SFR_IO_ADDR(DDRB), 1 // MMC SCLK pin (DDR, PORT)
> #define PORT_CK _SFR_IO_ADDR(PORTB), 1
> #define DDR_CK _SFR_IO_ADDR(DDRA), 6 // MMC SCLK pin (DDR, PORT)
> #define PORT_CK _SFR_IO_ADDR(PORTA), 6
> So it seems that the SPI protocol was redefined in asmfunc.S and you can
> change any pin you want (as soon as the I/O register addresses are between

cool I didin't know that.
so it works now ?

kernel64bits, 2019-01-28:
Still not with my pins :/ I want to use pins 7, 47,48,49 which are on port H and L.
With these pins the code won’t compile since sbi, cbi and sbic instructions are available only for registers which address between 0 and 31. 

I need to get rid of those intructions but I didn’t manage to replace them yet.

gerdmestdagh, 2019-01-28:
but you can get it to work without pin50-53 using those lower pins?

kernel64bits, 2019-01-29:
Yes but I am using a PCB where I plug all the components: changing the SD card reader pins would require to design a new PCB which I would like to avoid.

kernel64bits, 2019-02-16:
Ok, I found out what went wrong: the register 16 is already being used by the bootloader for other purposes so I was creating interferences when using it. My code is working fine with R18 :)

Here's a modified version of asmfunc.S that should be working with any pin:
[asmfunc.S.txt](https://github.com/FleetProbe/MicroBridge-Arduino-ATMega2560/files/2872060/asmfunc.S.txt)

closed 2019-02-16 by kernel64bits