FleetProbe/MicroBridge-Arduino-ATMega2560 #19: Change of EEPROM address for update check

aviatorhh, 2022-09-26:
I have changed the CS pin in asmfunc.S to get it running on the Ethernet Shield 2 where the CS pin for the SD access is 4 (PG5). I am happy about your work and the update process is functioning. But how to change the EEPROM address (0x1ff) for the use of the flags 0xf0 and 0xff? I am using the EEPROM to store data which shall persist updates. Or is the EEPROM erased anyway?

Mesger1, 2022-09-26:
stk500boot.c needs to be compiled and loaded as bootloader into your atmega2560. This way you can adjust to your liking. There is a short discription on how to compile.

aviatorhh, 2022-09-26:
I did already (see CS pin change), but I just wanted to point out the (easy) change of the EEPROM address as there sometimes is data in the EE.
Thank you